PowerBroker2/DFPlayerMini_Fast #7: playFolder don´t play, but playNext work

adbensi, 2020-02-27:
Hello!
Please. I wold like to know if the instructions to make SD folders still valid at this form:

`
 Pasta de E:\01

26/02/2020  18:35    <DIR>          .
26/02/2020  18:35    <DIR>          ..
19/12/2016  20:25           126.967 0000.mp3
30/11/2016  09:38            86.738 0001.mp3
30/11/2016  09:42            91.426 0002.mp3
30/11/2016  09:46            78.960 0003.mp3
20/12/2016  18:58            99.307 0004.mp3
20/12/2016  19:01           172.487 0005.mp3
30/11/2016  12:43           167.056 0006.mp3
30/11/2016  09:44            87.566 0007.mp3
20/12/2016  19:02           153.182 0008.mp3
30/11/2016  10:09            90.980 0009.mp3
07/11/2016  14:18            93.207 0010.mp3
20/12/2016  19:05            93.740 0011.mp3
20/12/2016  19:06            81.500 0012.mp3
20/12/2016  19:09            68.231 0013.mp3
20/12/2016  19:11           107.460 0014.mp3
20/12/2016  19:16            72.587 0015.mp3
20/12/2016  19:15           114.659 0016.mp3
31/12/2016  19:58           283.143 0017.mp3
12/01/2017  13:46           166.137 0018.mp3
22/12/2016  20:36           514.161 0019.mp3
              20 arquivo(s)      2.749.494 bytes
               2 pasta(s)    7.902.806.016 bytes disponíveis
`

because, this Works,
`
//***********************************************
SoftwareSerial    mySerial(-1, 2);
//***********************************************
//****************************************************************************/
void setup()
//****************************************************************************/
{
 //*********************************************** 
  mySerial.begin(9600);
  myDFPlayer.begin(mySerial);
  //***********************************************
  //***********************************************
  myDFPlayer.playNext();
  delay(2000);
  //***********************************************
`

but, this is ignored, and don´t play..
`
//***********************************************
SoftwareSerial    mySerial(-1, 2);
//***********************************************
//****************************************************************************/
void setup()
//****************************************************************************/
{
 //*********************************************** 
  mySerial.begin(9600);
  myDFPlayer.begin(mySerial);
  //***********************************************
  //***********************************************
  myDFPlayer.playFolder(1, 0);
  delay(2000);
  //***********************************************
`
I have ESP8266 and last lib. Please, what may be wrong here?
I am use 0000.mp3 and folder 01 names..

PowerBroker2, 2020-02-27:
`SoftwareSerial mySerial(-1, 2);`

I know this probably isn't the source of the problem, but why are one of your pins defined as -1? Never heard of pin -1...

Regardless, I'll take a look into it and let you know what I find. I'm having trouble understanding the contents of your SD card. Are all those mp3 files in root, in a custom named folder in root, in a folder named "mp3" in root, or a folder in a folder named "mp3" in root? As in:

1.) /0000.mp3
2.) /<custom folder name>/0000.mp3
3.) /mp3/0000.mp3
4.) /mp3/<custom folder name>/0000.mp3

Which one of the above accurately describes how you have your SD card setup?

PowerBroker2, 2020-02-27:
Nvm, I see you have a folder in root named "1"

PowerBroker2, 2020-02-27:
Does the SD card have any other folders in root other than "1"?

adbensi, 2020-02-27:
Hello PowerBroker2!

 I have others folders, 01 to 09,
 and all files start from 0000.mp3 to 0099.mp3

 I use only one TX pin on the SoftSerial, I don`t have RX in there, then I use -1 to configure it. When SoftSerial get -1the RX setup is ignored.

 This SD card is FAT32 8GB, I try others too.. but, witch playNext() I play it.. then, this is the problem.. I can configure volume, playNext(), but.. not play a especific file in the folder.

```CPP
 Pasta de E:\

11/11/2019  16:32    <DIR>          01
11/11/2019  16:32    <DIR>          02
11/11/2019  16:32    <DIR>          03
11/11/2019  16:32    <DIR>          04
11/11/2019  16:32    <DIR>          05
11/11/2019  16:32    <DIR>          06
11/11/2019  16:32    <DIR>          07
11/11/2019  16:32    <DIR>          08
11/11/2019  16:32    <DIR>          09
               0 arquivo(s)              0 bytes
               9 pasta(s)    7.902.806.016 bytes disponíveis
```

```CPP
C:\Users\Alexandre>dir E:\01
 O volume na unidade E não tem nome.
 O Número de Série do Volume é 1A06-1AD4

 Pasta de E:\01

26/02/2020  18:35    <DIR>          .
26/02/2020  18:35    <DIR>          ..
19/12/2016  20:25           126.967 0000.mp3
30/11/2016  09:38            86.738 0001.mp3
30/11/2016  09:42            91.426 0002.mp3
30/11/2016  09:46            78.960 0003.mp3
20/12/2016  18:58            99.307 0004.mp3
20/12/2016  19:01           172.487 0005.mp3
30/11/2016  12:43           167.056 0006.mp3
30/11/2016  09:44            87.566 0007.mp3
20/12/2016  19:02           153.182 0008.mp3
30/11/2016  10:09            90.980 0009.mp3
07/11/2016  14:18            93.207 0010.mp3
20/12/2016  19:05            93.740 0011.mp3
20/12/2016  19:06            81.500 0012.mp3
20/12/2016  19:09            68.231 0013.mp3
20/12/2016  19:11           107.460 0014.mp3
20/12/2016  19:16            72.587 0015.mp3
20/12/2016  19:15           114.659 0016.mp3
31/12/2016  19:58           283.143 0017.mp3
12/01/2017  13:46           166.137 0018.mp3
22/12/2016  20:36           514.161 0019.mp3
              20 arquivo(s)      2.749.494 bytes
               2 pasta(s)    7.902.806.016 bytes disponíveis

C:\Users\Alexandre>dir E:\02
 O volume na unidade E não tem nome.
 O Número de Série do Volume é 1A06-1AD4

 Pasta de E:\02

26/02/2020  18:35    <DIR>          .
26/02/2020  18:35    <DIR>          ..
12/01/2017  22:00            54.627 0000.mp3
12/01/2017  11:39             5.421 0001.mp3
12/01/2017  11:41             9.591 0002.mp3
12/01/2017  12:32             8.340 0003.mp3
12/01/2017  12:32             8.340 0004.mp3
12/01/2017  12:33            10.842 0005.mp3
12/01/2017  12:34            12.093 0006.mp3
12/01/2017  12:35            10.425 0007.mp3
12/01/2017  12:36             8.757 0008.mp3
12/01/2017  12:36            10.425 0009.mp3
12/01/2017  12:37            10.008 0010.mp3
12/01/2017  12:37             9.174 0011.mp3
12/01/2017  12:40             9.591 0012.mp3
12/01/2017  12:40             7.923 0013.mp3
12/01/2017  12:41            11.676 0014.mp3
12/01/2017  12:41             8.340 0015.mp3
12/01/2017  12:42            15.846 0016.mp3
12/01/2017  12:42            15.429 0017.mp3
12/01/2017  12:44            11.676 0018.mp3
12/01/2017  12:43            12.510 0019.mp3
12/01/2017  12:44             9.591 0020.mp3
12/01/2017  12:45            11.676 0021.mp3
12/01/2017  12:45            16.263 0022.mp3
12/01/2017  12:46            17.514 0023.mp3
12/01/2017  11:37            12.093 0024.mp3
12/01/2017  12:47            63.737 0025.mp3
12/01/2017  12:48            55.378 0026.mp3
12/01/2017  12:48            60.603 0027.mp3
12/01/2017  12:48            68.962 0028.mp3
12/01/2017  12:53            10.008 0029.mp3
12/01/2017  12:54             6.255 0030.mp3
12/01/2017  12:55            10.842 0031.mp3
12/01/2017  12:56            18.348 0032.mp3
12/01/2017  12:57            17.514 0033.mp3
12/01/2017  12:57            15.012 0034.mp3
12/01/2017  12:58            17.097 0035.mp3
12/01/2017  12:58            18.765 0036.mp3
12/01/2017  12:59            17.514 0037.mp3
12/01/2017  12:59            18.348 0038.mp3
12/01/2017  13:00            15.429 0039.mp3
12/01/2017  13:00             8.340 0040.mp3
12/01/2017  13:01            11.676 0041.mp3
12/01/2017  13:08            15.012 0042.mp3
12/01/2017  13:09            14.595 0043.mp3
12/01/2017  13:09            16.263 0044.mp3
12/01/2017  13:10            16.263 0045.mp3
12/01/2017  13:10            16.263 0046.mp3
12/01/2017  13:11            15.429 0047.mp3
12/01/2017  13:11            15.012 0048.mp3
12/01/2017  13:11            14.178 0049.mp3
12/01/2017  13:12            10.008 0050.mp3
12/01/2017  13:12            12.927 0051.mp3
12/01/2017  13:13            16.263 0052.mp3
12/01/2017  13:13            16.680 0053.mp3
12/01/2017  13:14            17.097 0054.mp3
12/01/2017  13:14            18.348 0055.mp3
12/01/2017  13:15            18.348 0056.mp3
12/01/2017  13:15            18.348 0057.mp3
12/01/2017  13:16            16.263 0058.mp3
12/01/2017  13:16            15.012 0059.mp3
12/01/2017  13:24            13.344 0060.mp3
12/01/2017  13:25            16.263 0061.mp3
12/01/2017  13:33            20.433 0062.mp3
12/01/2017  13:26            19.599 0063.mp3
12/01/2017  13:26            18.348 0064.mp3
12/01/2017  13:27            19.599 0065.mp3
12/01/2017  13:27            20.433 0066.mp3
12/01/2017  13:27            19.599 0067.mp3
12/01/2017  13:28            18.348 0068.mp3
12/01/2017  13:29            17.931 0069.mp3
              70 arquivo(s)      1.218.205 bytes
               2 pasta(s)    7.902.806.016 bytes disponíveis
```

adbensi, 2020-02-27:
I put an mp3 folder with a 0000.mp3 file, and the call playFromMP3Folder(0) works.

I'll print what is being sent to the module DFPlayerMini when playfolder(1,0) is call:
```CPP
7E FF 06 0F 00 01 00 FE EB EF
```

and when mp3folder is call:
```CPP
7E FF 06 12 00 00 00 FE E9 EF 
```

adbensi, 2020-02-27:
I have news,
I changed the names files from 00XX.mp3 to 0XX.mp3, and it play from 001.mp3 file now.
I can´t play 000.mp3 file from playfolder(1,0) function. I have issues to solve yet..

Then, I rename file 000.mp3 to 020.mp3 and it play now. May be wrong CRC or hardware issues?
I try others 4 modules.. same problem..

I need to talk, on the datasheet the command 0x14 tell to use XXXX names, but this lib use command 0x0F to use XXX names, on the WIKI it tell to use XXXX.mp3 format. May be I need to change command 0x0F to 0x14, but I have other question to investigate.. may be I have another problem, please, did You can help me to investigate why I can not play 000 names?

What I can do to debug it? 
Because I think I may have some names that will not be reproduced or fail at times.

adbensi, 2020-02-27:
Hi again.. sorry about that.. I need to use XXXX format file names.. then, I change the code, but, not work.. please, did You can help me?

I test this without sucess:
```CPP
void DFPlayerMini_Fast::playFolder(uint8_t folderNum, uint8_t trackNum)
{
	//sendStack.commandValue  = SPEC_FOLDER;
	sendStack.commandValue  = SPEC_TRACK_3000;
	sendStack.feedbackValue = NO_FEEDBACK;
	//sendStack.paramMSB = folderNum;
	//sendStack.paramLSB = trackNum;
	
	uint16_t TRACK_3000 = ((((uint16_t)folderNum) << 12) | trackNum);
	sendStack.paramMSB = ((TRACK_3000 & 0xFF00) >>8);
	sendStack.paramLSB = (TRACK_3000 & 0x00FF);

	findChecksum(&sendStack);
	sendData();
}
```

PowerBroker2, 2020-02-27:
Assuming you're using folders "1"-"9" in root and mp3 filenames in each folder start with index of 1, playFolder(folder, track) will correctly play "track" in "folder". If you're trying to play all files within a folder, use repeatFolder(folder).

Please note that track number 0 is invalid. Track numbers begin numbering at 1. Also note that both folder and track numbers are dependent only on creation time and NOT folder/track names. Folder/track names with numbers are only to help the user/programmer understand which mp3 files will be called given a DFPlayer command. The DFPlayer disregards folder/filenames.

For further information, please refer to the documentation PDF within "extras" inside this repository and the DFPlayerMini's Wiki.

Can you confirm everything is working normal on your end?

adbensi, 2020-03-02:
Thank you for your help! 

Everything would be working now if I could use names in this order:
0001.mp3
0002.mp3

rather than
001.mp3
002.mp3

But this is not a problem in the library, thanks to the explanations.

I will however implement the function for long names 0001, the function below should work, but something is still missing with the findChecksum, and I don't know what it is yet.

```CPP
void DFPlayerMini_Fast::playFolderExtended(uint8_t folderNum, uint8_t trackNum)
{
	sendStack.commandValue  = SPEC_TRACK_3000;
	sendStack.feedbackValue = NO_FEEDBACK;
	
	uint16_t TRACK_3000 = ((((uint16_t)folderNum) << 12) | trackNum);
	sendStack.paramMSB = ((TRACK_3000 & 0xFF00) >>8);
	sendStack.paramLSB = (TRACK_3000 & 0x00FF);

	findChecksum(&sendStack);
	sendData();
}
```

PowerBroker2, 2020-03-03:
1.) The library's checksum function doesn't have a problem, I think the problem is how you're forming the packets
2.) 7E FF 06 0F 00 01 14 -- -- EF is the correct packet (minus the checksum - you'll have to calculate that) for playing track 1 in folder 12.
3.) You don't have to use delayMicroseconds() between each write()
4.) If you want to send custom packets for testing, you should change the values in dfplayer.sendStack, call dfplayer.sendData(), call dfplayer.parseFeedback(), then figure out the response by calling dfplayer.printStack(dfplayer.recStack)

For reference:
```C++
	struct stack {
		uint8_t start_byte;
		uint8_t version;
		uint8_t length;
		uint8_t commandValue;
		uint8_t feedbackValue;
		uint8_t paramMSB;
		uint8_t paramLSB;
		uint8_t checksumMSB;
		uint8_t checksumLSB;
		uint8_t end_byte;
	} sendStack, recStack;
```

closed 2020-03-04 by PowerBroker2