PowerBroker2/DFPlayerMini_Fast #8: Problem compiling for the nano 33 ble

derekchall, 2020-04-16:
Compiling for the nano 33 ble gives an error when GET_STATUS is used, this seems to be because GET_STATUS is defined some ware in the Arduino mbed stuff, it can be fixed by changing it to GET_Satus in your library.

PowerBroker2, 2020-04-16:
Thanks for bringing this to my attention. I'll see if I can't implement a library namespace to prevent naming conflicts like this.

PowerBroker2, 2020-04-16:
Should be fixed in release 1.1.4

PowerBroker2, 2020-04-16:
Please reopen if you have further problems related to this issue. Thanks again!

derekchall, 2020-04-17:
Still not working, looks like the namespace dose not override the #define statement

```
> C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/USB/USBCDC.h:23:0,

                 from C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/USB/PluggableUSBSerial.h:22,
                 from C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Arduino.h:96,
                 from C:\Users\derek\AppData\Local\Temp\arduino_build_153490\sketch\DFPlayer_mini.ino.cpp:1:
C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/drivers/internal/USBDevice_Types.h:22:28: error: expected unqualified-id before numeric constant
 #define GET_STATUS        (0)
                            ^
C:\Users\derek\Documents\Arduino\libraries\DFPlayerMini_Fast\src/DFPlayerMini_Fast.h:55:16: note: in expansion of macro 'GET_STATUS'
  const uint8_t GET_STATUS       = 0x42;
                ^~~~~~~~~~
C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/drivers/internal/USBDevice_Types.h:22:28: error: expected ')' before numeric constant
 #define GET_STATUS        (0)
                            ^
C:\Users\derek\Documents\Arduino\libraries\DFPlayerMini_Fast\src/DFPlayerMini_Fast.h:55:16: note: in expansion of macro 'GET_STATUS'
  const uint8_t GET_STATUS       = 0x42;
                ^~~~~~~~~~
Using library DFPlayerMini_Fast at version 1.1.4 in folder: C:\Users\derek\Documents\Arduino\libraries\DFPlayerMini_Fast 
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

PowerBroker2, 2020-04-17:
Ok, should be fixed with 1.1.5

closed 2020-04-17 by PowerBroker2

derekchall, 2020-04-17:
Thanks, it’s all working now, thanks for a very useful library. If you are interested is what I will be doing with this library see my YouTube channel, Thanks again
https://www.youtube.com/micromouse