PowerBroker2/DFPlayerMini_Fast #10: function STOP

Roman-RV78, 2020-04-18:
Hello. The library does not have the STOP command, the variable is declared const uint8_t STOP = 0x16;, and the function is not present.
EQ Values I think I need to take out for the namespace dfplayer block, otherwise for example the command myMP3.EQ Select(EQ_NORMAL); not declared

PowerBroker2, 2020-04-18:
I'll take a look at the STOP command issue, but you can probably just use PAUSE.

As for the EQ_NORMAL not declared issue, I added the namespace `dfplayer` to possibly fix issue #8. In order to use constants such as this, you will need to prepend `dfplayer::` to the constants: `dfplayer::EQ_NORMAL`

PowerBroker2, 2020-04-20:
Added stop() function in [release 1.1.6](https://github.com/PowerBroker2/DFPlayerMini_Fast/releases/tag/1.1.6)

closed 2020-04-20 by PowerBroker2