PowerBroker2/DFPlayerMini_Fast #12: Play files in a folder randomly

Tsjippy, 2020-04-29:
Thanks for this library!

Is it possible to randomly play the content of a specific folder?
I currently have this, but maybe there is an easier way:
`
#include "Arduino.h"
#include "SoftwareSerial.h"
#include <DFPlayerMini_Fast.h>

int currentFolder = 0;

SoftwareSerial mySoftwareSerial(10, 11); // RX, TX
DFPlayerMini_Fast MP3_player;

void setup(){
  mySoftwareSerial.begin(9600);
  Serial.begin(115200);

  //Initialize a a randomseed for the random number generator
  randomSeed(analogRead(0));
}

void loop(){
  playFolder(6);

  if(MP3_player.isPlaying()){
    Serial.print("Currently playing track");
    Serial.println(MP3_player.currentSdTrack());
  }else if (currentFolder != 0){
    //If playing is over and playing a folder, play the next random track
    playFolder(currentFolder);
  }
  delay(1000);
}

void playFolder(int folder){
  currentFolder = folder;
  int folderSize = MP3_player.numTracksInFolder(folder);
  MP3_player.playFolder(folder, random(1,folderSize+1));
}
`

Tsjippy, 2020-04-29:
`test`

PowerBroker2, 2020-04-29:
Having properly formatted code helps:

```C++
#include "Arduino.h"
#include "SoftwareSerial.h"
#include <DFPlayerMini_Fast.h>

SoftwareSerial mySoftwareSerial(10, 11); // RX, TX
DFPlayerMini_Fast MP3_player;

int currentFolder = 0;

void setup()
{
  Serial.begin(115200);
  mySoftwareSerial.begin(9600);
  MP3_player.begin(mySoftwareSerial);
  
  randomSeed(analogRead(0));
}

void loop()
{
  playFolder(6);
  
  if (MP3_player.isPlaying())
  {
    Serial.print("Currently playing track");
    Serial.println(MP3_player.currentSdTrack());
  }
  else if (currentFolder != 0)
  {
    //If playing is over and playing a folder, play the next random track
    playFolder(currentFolder);
  }
  
  delay(1000);
}

void playFolder(int folder)
{
  currentFolder = folder;
  
  int folderSize = MP3_player.numTracksInFolder(folder);
  MP3_player.playFolder(folder, random(1, folderSize + 1));
}
```

That being said, I don't think the DFPlayerMini has support for a specific command that automatically plays all files in a folder at random. You can, however, call `MP3_player.randomAll()`, but this of course isn't limited to only the files in a single folder.

closed 2020-04-29 by PowerBroker2