PowerBroker2/DFPlayerMini_Fast #13: Is there a way to play certain tracks?

Chris-Blackwell, 2020-07-08:
The idea is that I am putting several albums by one artist in a file. I would like to start at the first track of the album and end on the last. I was looking at playFolder, but I am not sure if this would work for my implementation. So I would like to start with folder 1, 001,mp3 and play through to 12. But maybe later I want to listen to the second album so I want to play folder 1 starting at 0013.mp3 and listen through the next 11 songs. 

Fantastic job with the library by the way!

PowerBroker2, 2020-07-08:
Thank you!

In terms of your question, you would use
```c++
void playFolder(uint8_t folderNum, uint8_t trackNum);
```
to play a given track in a given folder. Keeping track of what folder number corresponds to what album and what track number corresponds to what song will be up to you since there's no way to query folder/track metadata. You could create a const array in program memory to serve as an album/track lookup table (or something like it) to keep track of what corresponds to what.

Does that help?

Chris-Blackwell, 2020-07-08:
Thank you for the reply. I did implement pulling the folder and the file from the rfid card, but I am having trouble with how to play the next x number of songs. I have a third variable on the card which is the number of tracks on the album. The method isPlaying() is always returning as 1 for me. I am just having trouble trying to figure out how to loop through the tracks i want to play. I have something like the following loop (pseudocode):
`while (track < albumCount){
    value = mp3.isPlaying();
     if (value == false){
         delay(1000);
          mp3.playFolder(folder,track);
          value = mp3.isPlaying();
           track++;
           }else{
              delay(1000);
  } `
Thanks again for any feedback.

PowerBroker2, 2020-07-08:
It would be better to poll the DFPlayerMini's busy pin instead of using the `isPlaying()` method. You can do something like this:
```c++
while (track < albumCount)
{
  if (!digitalRead(BUSY_PIN))
  {
    mp3.playFolder(folder, track);
    track++;
  }
}
```
where `BUSY_PIN` is defined by the digital input pin on the Arduino connected to the busy pin of the DFPlayerMini.

Chris-Blackwell, 2020-07-08:
LOL I literally just implemented that right before you posted that! I am cycling through the tracks right now. Great minds! (Not saying I have a great mind though, lol). Thank you so much for your help.

PowerBroker2, 2020-07-08:
lmao, glad it's working!

closed 2020-07-08 by PowerBroker2