PowerBroker2/DFPlayerMini_Fast #14: Using the busy pin never seems to work

MikeS11, 2020-08-03:
Not too sure whats going on but I've had this working before on another project but right now it seems to be buggy.

Simply trying reading in PIN_16 or the busy pin to check to see if anything is currently playing before starting an idle track just keeps trying to put the idle track in an endless loop. When I check the serial inputs I see the correct inputs when something is playing but when I run this simple code to check I see inputs flippings from 0,1 over and over again. I figure play this track if there is nothing else playing and it should work... Is there something I am missing?

Thanks!

        if (digitalRead(ISPLAYING_OUT) == true)
        {
          myDFPlayer.loop(idleTrack);
        }

PowerBroker2, 2020-08-03:
Please post your entire code and what exactly is going wrong. If it's a library issue I can help, but not if it's a hardware issue

MikeS11, 2020-08-04:
I cant post my whole code but this is roughly it:

/*********************************************************
    Initiate Libraries for DFPlayer sound module
 *********************************************************/

#include "SoftwareSerial.h"
#include <DFPlayerMini_Fast.h>

SoftwareSerial mySoftwareSerial(10, 11); // RX, TX
DFPlayerMini_Fast myDFPlayer;

// Audio Tracks - Note DFPlayer is sensitive to file order
unsigned long  startupTrack = 1;
unsigned long  blastTrack = 2;
unsigned long  endTrack = 4;
unsigned long  shutdownTrack = 10;
unsigned long  idleTrack = 5;
unsigned long  clickTrack = 6;
unsigned long  chargeTrack = 7;
unsigned long  ventTrack = 8;
unsigned long  warnTrack = 9;
unsigned long  choreTrack = 1;

void setup()

  mySoftwareSerial.begin(9600);
  if (!myDFPlayer.begin(mySoftwareSerial))
  {
    while (true);
  }

void loop()
///// a bunch of routines that could be playing sound effects,  and this idle sound below to run when everything stops:

if (digitalRead(ISPLAYING_OUT) == true)
    {
      myDFPlayer.loop(idleTrack);
    }

THis is a DFPlayer I got from Amazon in a batch of 5 of them I think from China.... I just ordered an actual DFPlayer from Digikey as well that I will test in a couple days... I think these knockoff ones are garbage

PowerBroker2, 2020-08-04:
Please post the entire code. It's either a problem with your sketch, the wiring, or the lib, so I need to see all the software you're using

MikeS11, 2020-08-05:
Looks like it was an issue with the DFPlayer being a garbage knock off, the new one from digikey works like a boss.

MikeS11, 2020-08-09:
Sorry I had to reopen:

I am having a problem where I have three states of audio, one is a start up track then idle track then off track,

Looks like the playfolder() routine is hanging the first play from happening: 

If I remove the play folder, the timing of the on and off sounds are perfect, when I keep the line, the first sound does not play or only does on certain occasions.E.g. it will always play if i turn off the switch then on while the off sound is playing, but from a truly off state the start sound gets buggered and the start sound never plays. 

Its driving me bananas.

#include <SoftwareSerial.h>
#include <DFPlayerMini_Fast.h>

SoftwareSerial mySoftwareSerial(10, 11); // RX, TX
DFPlayerMini_Fast myMP3;

const int STARTWAND_SWITCH = 6;
const int ISPLAYING_OUT = 12;

// Audio Tracks - Note DFPlayer is sensitive to file order
unsigned long  startupTrack = 1;
unsigned long  blastTrack = 2;
unsigned long  themeTrack = 3;
unsigned long  endTrack = 4;
unsigned long  clickTrack = 6;
unsigned long  chargeTrack = 7;
unsigned long  ventTrack = 8;
unsigned long  warnTrack = 9;
unsigned long  shutdownTrack = 10;

bool off = true;
bool startwand = false;
bool Notplaying = true;
bool isIdling = false;

unsigned long  FireMovieTrack = 2;
unsigned long  FireStasisTrack = 3;
unsigned long  FireSlimeTrack = 4;
unsigned long  FireMesonTrack = 5;
unsigned long  FireMovie2Track = 6;
unsigned long  IdleMovieTrack = 7;

void setup()
{
  Serial.begin(9600);

  mySoftwareSerial.begin(9600);
  if (!myMP3.begin(mySoftwareSerial)) {  //Use softwareSerial to communicate with mp3.
    while (true);
  }
  myMP3.volume(30);

  pinMode(STARTWAND_SWITCH, INPUT);
  digitalWrite(STARTWAND_SWITCH, HIGH);
  pinMode(ISPLAYING_OUT, INPUT);
  digitalWrite(ISPLAYING_OUT, LOW);
}

void loop()
{
  startwand =  digitalRead(STARTWAND_SWITCH);
  

  if (startwand == true)
  {
    //Serial.println(off);
    if (off == true)
    {
      myMP3.play(1);
      off = false;
    }
    Notplaying = digitalRead(ISPLAYING_OUT);
    if (Notplaying == true && isIdling == false)
    {
      Serial.println(isIdling);
      isIdling = true;
      myMP3.playFolder(7,2);
    }
  }
  else
  {
    if (off == false)
    {
      off = true;
      isIdling = false;
      myMP3.play(4);
    }
  }
  delay(1);
}

MikeS11, 2020-08-12:
Let me know if this is just a limitation with the hardware or if the library can be improved.

PowerBroker2, 2020-08-17:
Your specific application is hard for me to understand. If `myMP3.playFolder()` works in a simple example but not your project code, it is your project code that needs to be fixed. Note that you should have at least 10ms delay between all DFPlayer calls (commands and queries alike)

closed 2020-08-29 by PowerBroker2