PowerBroker2/DFPlayerMini_Fast #15: query() not working

hansliss, 2020-08-31:
At least on the noname DFPlayer I have, the query() method always returns -1. I found two problems: 

1. The recChar variable in parseFeedback() needs to be an uint8_t (or unsigned char, obviously) for the state machine to get anywhere since VER=0xFF
2. The 'cmd' parameter in the call to getStatus() within query() needs to be dfplayer::REPLY - clearly, the received response stack uses 0x41 as the cmd value.

PowerBroker2, 2020-08-31:
Interesting, I'll take a look into it soon

PowerBroker2, 2020-09-01:
You're right about the `recChar` point, but not the `cmd` one since the value of `cmd` depends on what query you put to the MP3 player. I've changed the `recChar` variable to uint8_t - let me know if it fixes things for you.

hansliss, 2020-09-01:
Sadly, no. On my unit, at least the GET_STATUS_, GET_VOL and GET_U_FILES return a 0x41 reply, so they don't work unless I change row 608 to 
	if (getStatus(dfplayer::REPLY))
I take it these queries work on others' units? That means there are different versions out there with slightly different behaviour.

hansliss, 2020-09-01:
I'll make a pull request with a non-breaking change, adding a "protocol mode" as an optional parameter. It's up to you whether you want to include it or not.
Incidentally, my unit seems to need the "threshold" lowered to 200 to work reliably, too.

PowerBroker2, 2020-09-01:
Yeah, I was messing around with Travis CI, but unfortunately I couldn't get it to work across all Arduino boards (surprisingly not all boards support SoftwareSerial.h for some reason), so don't worry about getting the builds to go through.

I'll do the change myself - didn't realize the command parameter changed during query responses.

PowerBroker2, 2020-09-01:
Ok, try it now

hansliss, 2020-09-01:
Will the change you made not break this for the devices that must have worked previously? That's why I made it a configuration parameter in my fork, to make it possible to get it to work with both types of device.
I've sent a message to the manufacturer to try to get confirmation that there are multiple versions of the DFPlayer Mini with subtly different behaviour. I'll let you know if they respond. We should keep this issue open for now, I think.

PowerBroker2, 2020-09-01:
Ok, I kinda took your word for it, but now I'm looking at the datasheet [here](https://github.com/PowerBroker2/DFPlayerMini_Fast/blob/master/extras/FN-M16P%2BEmbedded%2BMP3%2BAudio%2BModule%2BDatasheet.pdf) and I think what I had before was correct. Take a look at the table at the top of page 8. It seems the cmd parameter is supposed to be the same for both query and response. Idk why your module would act any different.

hansliss, 2020-09-01:
It's an MP3-TF-16P, so not exactly the same model. But yes, that's why I made it configurable, and why I contacted the original manufacturer. Take a look at https://github.com/hansliss/DFPlayerMini_Fast/src and you'll see how I did it.

PowerBroker2, 2020-09-01:
Ok, interesting, I'll make it configurable. I really appreciate your help in identifying these issues and helping fix them!

hansliss, 2020-09-01:
Happy to help!  The manual you had was new to me, btw - here's the manual I've been reading: http://image.dfrobot.com/image/data/DFR0299/DFPlayer%20Mini%20Manul.pdf
Interestingly, this one lists 0x41 as "Reply", and doesn't have that detailed table with query/response data.

hansliss, 2020-09-01:
Oh dear. Sorry to spam you, but I just re-checked page 8 in your datasheet, and it turns out the columns are *not* "sent data" and "received data" but "sent data with checksum" and "sent data without checksum" - check the column headers! Section "3.5.3 Returned data of feedback from module" suggests vaguely that 0x41 is in fact the response code.

PowerBroker2, 2020-09-01:
Gotta love horribly written documentation - good catch. Can you test the current master branch? If it works I'll just tag out a new version real quick

hansliss, 2020-09-01:
Yes, the change works for me!

closed 2020-09-01 by PowerBroker2