PowerBroker2/DFPlayerMini_Fast #18: Get list of files or count files

T-vK, 2020-09-04:
I have an Arduino with two buttons connected. The first button selects a file and the second button plays the currently selected file.

At the moment the button increases the value of a variable whenever it is pressed. For example:

``` C++
int selectedSong = 1;

selectSongButtonState = digitalRead(selectSongButton);
if (selectSongButtonState == HIGH) {
  selectedSong++;
}

playSongButtonState = digitalRead(playSongButton);
if (playSongButtonState == HIGH) {
  dfPlayer.play(selectedSong);
}
```

But I don't know how many files are stored on the SD card. So let's say there are only 3 files. In that case, whenever the first button is pressed, the `selectedSong` variable needs to be increased until it reaches a value of 3 then it needs to start at 1 again. 

Is there a way to find out how many files there currently are?

PowerBroker2, 2020-09-04:
You should be able to use `numSdTracks()` to get the number of files on the SD card

T-vK, 2020-09-04:
Brilliant! Thank you so much!

closed 2020-09-04 by PowerBroker2

T-vK, 2020-09-19:
I just realized that this doesn't work. It always returned -1.

My files are in the root directory of the SD card btw and they are called `0001.mp3`, `0002.mp3`, `0003.mp3` and `0004.mp3`

I also tried creating a folder called 01 and copied those files in there. Still with the same result. Then I tried using `numTracksInFolder(1)` instead, but still, I get -1.

Any ideas how I could get this to work?

PowerBroker2, 2020-09-19:
Ensure your wiring is correct AND that you're using the latest version of the lib

T-vK, 2020-09-19:
I've used 1.1.9 which seems to be the latest version. The wireing should be fine as I can play files just fine. I'm using a 1k resistor and tx and another one on rx. I have the whole thing soldered onto a PCB already btw.

T-vK, 2020-09-22:
Is there any chance you can verify that this really works and there is no bug in the code?

T-vK, 2020-09-28:
Still haven't been able to figure it out. :/

PowerBroker2, 2020-09-30:
Try [release 1.1.10](https://github.com/PowerBroker2/DFPlayerMini_Fast/releases/tag/1.1.10)

T-vK, 2020-10-02:
I've tried it and now I always get a count of 0, not matter if I put the files in the root directory or if I create a folder called `01` and place them in there. 

Here is a simple way to reproduce:

``` C++
#include <SoftwareSerial.h>
#include <DFPlayerMini_Fast.h>

const int DF_PLAYER_RX_PIN = 9;
const int DF_PLAYER_TX_PIN = 8;

SoftwareSerial mp3PlayerSerial(DF_PLAYER_RX_PIN, DF_PLAYER_TX_PIN);
DFPlayerMini_Fast mp3Player;

int totalTrackCount;

void setup()
{
  Serial.begin(9600);
  delay(500);
  mp3PlayerSerial.begin(9600);
  delay(500);
  mp3Player.begin(mp3PlayerSerial);
  delay(500);
  totalTrackCount = mp3Player.numTracksInFolder(1);
  Serial.print("numTracksInFolder: ");
  Serial.println(totalTrackCount);
  delay(500);
  totalTrackCount = mp3Player.numSdTracks();
  Serial.print("numSdTracks: ");
  Serial.println(totalTrackCount);
}

void loop()
{
  //do nothing
}
```

It always reports 

```
numTracksInFolder: 0
numSdTracks: 0
```

for me on my Arduino Pro Micro.

But there definitely are 4 files on my micro SD card (0001.mp3, 0002.mp3, 0003.mp3 and 0004.mp3). There is nothing else on that card and as I said playing any of these files using your library works just fine.

aviost, 2021-01-09:
Hi, I see this thread is closed but is there a solution for this iasue?

T-vK, 2021-01-09:
Nope.

PowerBroker2, 2021-01-16:
See [this](https://github.com/PowerBroker2/DFPlayerMini_Fast/issues/26#issuecomment-760194900) - perhaps it provides some insight into this issue as well. I'd be happy to fix the issue if it really is a lib bug, but based of the above link, I think the real issue is your board's firmware. If you have reason to believe otherwise, please give me the details and I'll fix it.

T-vK, 2021-01-19:
Interesting, so it's probably that my DFPlayer modules have a chip with less features.
I wasn't aware there are so many different chips.

I would really like to see a mention of supported chips in the Readme.

The situation seems to be completely insane. Here is what I found on Aliexpress (and I'm sure there is more):

- MH2024K-24SS
- YX5200-24SS
- BY8001-24SS
- AA20HFJ616-94
- DFROBOT LISP3
- AA1750CJ????.1-94 
- AA1828CJ5M2U.1-94
- ??????????????

(MH-ET LIVE) MH2024K-24SS
https://www.aliexpress.com/item/32271894202.html
[image]

YX5200-24SS
https://www.aliexpress.com/item/32860342792.html
[image]

BY8001-24SS
https://www.aliexpress.com/item/33026303719.html
[image]

DFROBOT LISP3
https://www.aliexpress.com/item/32992229288.html
[image]

AA1750CJ????.1-94 
https://www.aliexpress.com/item/1005001986028946.html
[image]

AA1828CJ5M2U.1-94
https://www.aliexpress.com/item/4001066296677.html
[image]

??????????????
https://www.aliexpress.com/item/32673923788.html
[image]

PowerBroker2, 2021-01-19:
> I would really like to see a mention of supported chips in the Readme.

Once we get a final handle on what works and what doesn't work I can add the note or somebody could submit a PR