PowerBroker2/DFPlayerMini_Fast #19: Number of tracks in the "MP3" folder

kymodoke, 2020-09-11:
Hi,

Just like the function to get the number of tracks from a "numbered" folder ( `int16_t numTracksInFolder(uint8_t folder);` ), is there a way to get the number of tracks BUT from the folder named "MP3" or "mp3" ?
From documentation the "numTracksInFolder" can only use an integer from 0 to 255.

My intention is to play a random track from the "mp3" folder (and not from a numbered folder nor from the root)

example of usage:
```
void playRandomSongFromMp3Folder () {
long randomInt;
nbTracksMP3 = myMp3.numTracksInFolder('MP3'); // wil not work !
randomInt = random(nbTracksMP3);
myMp3.playFromMP3Folder(randomInt);
}
```

-> Is there a trick to use the "numTracksInFolder" function for that, or does it need to implement a new dedicated function ?

Thanks.

PowerBroker2, 2020-09-11:
If all of your tracks are only in `mp3` and not also in separate numbered folders, you can use `numSdTracks()` to get the information you're asking for

kymodoke, 2020-09-15:
@PowerBroker2 , thanks for your anwser.
Unfortunately I already use the numbered folders for another usage.

Context:
_My project is a speaking clock that tells the time every hours in different random manners (so folders 00, 01, 02,... 23 with several mp3 files for each hours) and but it also speaks randomly (over 300 files in folder mp3 randomized) several time per hour. It's inspired by the [Nabaztag project on RPi](https://github.com/nabaztag2018/pynab) (a connected rabbit that speaks)_ 

I've digged into the [DFPlayer datasheet](https://github.com/DFRobot/DFRobotDFPlayerMini/blob/master/doc/FN-M16P%2BEmbedded%2BMP3%2BAudio%2BModule%2BDatasheet.pdf)  if for the _**"Query number of tracks in a folder"**_ `7E FF 06 4E` I could use an hex code specific for the mp3 folder, but no... So I assume it is not possible...

For the moment in my code, I used a hard-coded/defined number of tracks for the MP3 folder, in order to do that randomization. But it would be better if not hard-coded but dynamically based on the real number of files in the mp3 folder, as the number of files could change.
-> I guess I'll have to go on a plan B to avoid hard-coded number : I would use a user defined variable from a web configuration portal ;)

By the way, the library is very handy and my function (based on numbered folders) that plays/speaks the hours does work well:

```
void NabPlayHour(int cur_hour) {
      long randomInt;
      uint8_t nbTracksHour;
    
      // Get the number of tracks in corresponding time folder
      nbTracksHour = myMP3.numTracksInFolder(cur_hour);

      // Get a random number from the folder track count
      randomInt = random(nbTracksHour);
    
      // Play Hours only withing the wake range
      if (NabSleeping == false) {
            myMP3.playFolder(cur_hour, randomInt);
      }
}
```

closed 2020-09-15 by PowerBroker2