PowerBroker2/DFPlayerMini_Fast #20: Functions did not work

MarphXL, 2020-09-29:
Can you please explain how I can get the Volume or Number of files at SD-Card? I ever get a 0. It would be superb if you add these lines to your example if the functions are working. Thanks for that library!

MarphXL, 2020-09-30:
I double-checked all connections and try the original DFRobot-library and this works. Switching back to DFPlayerMini_Fast and the readings from serial connection didn't work. Maybe a protocol-problem in your library?

PowerBroker2, 2020-09-30:
Try [release 1.1.10](https://github.com/PowerBroker2/DFPlayerMini_Fast/releases/tag/1.1.10)

MarphXL, 2020-10-04:
how I can get the Volume or Number of files at SD-Card?

PowerBroker2, 2020-10-04:
Why do you need to query for volume? Since your sketch sets the volume, it should be able to remember what level was set. The number of files can be found using `numSdTracks()`

closed 2020-10-18 by PowerBroker2