PowerBroker2/DFPlayerMini_Fast #22: Cannot compile when using ATTiny85

GeneArnold, 2020-11-06:
I was directed to this library because it does a better job with sleep. When giving it a try on an ATTiny85 I'm having an issue. Below is the error. Also, here is a link to someone with the same issue that PowerBroker2 responded to but the OP never returned a message.

https://www.reddit.com/r/arduino/comments/eeau2e/is_it_possible_to_use_the_dfplayermini_fast/

```
/Users/garnold/Documents/Arduino/libraries/DFPlayerMini_Fast-master/src/DFPlayerMini_Fast.cpp: In member function 'void DFPlayerMini_Fast::printStack(DFPlayerMini_Fast::stack)':
/Users/garnold/Documents/Arduino/libraries/DFPlayerMini_Fast-master/src/DFPlayerMini_Fast.cpp:1108:2: error: 'Serial' was not declared in this scope
  Serial.println("Stack:");
  ^~~~~~
/Users/garnold/Documents/Arduino/libraries/DFPlayerMini_Fast-master/src/DFPlayerMini_Fast.cpp:1108:2: note: suggested alternative: '_serial'
  Serial.println("Stack:");
  ^~~~~~
  _serial
exit status 1
Error compiling for board ATtiny25/45/85.
```

PowerBroker2, 2020-11-06:
Ah, I'm guessing the ATTiny85 doesn't have a hardware UART port and therefore doesn't have a defined `Serial` class instance? I assumed every Arduino board at least had that. I do know a couple of work arounds. You can comment out that function in both the cpp and header files and I can also edit and release a new version of the library where you specify a serial port instead of the lib assuming which one you want to use.

GeneArnold, 2020-11-06:
You are correct, no hardware serial on these little guys. That's what I
thought it was. I'll check out those files and see if I can figure it out.
I've not done that before.

PowerBroker2, 2020-11-06:
See [release 1.1.11](https://github.com/PowerBroker2/DFPlayerMini_Fast/releases/tag/1.1.11)

closed 2020-11-06 by PowerBroker2

fschlau, 2020-12-09:
Hi,
I still get an error _serial when compiling for the ATTiny85? 
I installed the 1.1.11.

Serial.println("Stack:");
  ^~~~~~
  _serial
exit status 1

GeneArnold, 2020-12-09:
I had to go into the source code and remove the serial calls that you
highlighted. The updated code does not fix the issue (at least from my
testing). I can say that after I made the change it does compile. The issue
I'm having now is just getting the darn ATTiny85 to produce serial
communications! Ugh, what a pain it has been. Even after following some
very good guides.

fschlau, 2020-12-09:
Okay, I just commented these print-statements out in the cpp File. Now it compiles, thank you.
I have the same hard road with the Attiny85, it just don't want to communicate...

fschlau, 2020-12-09:
If got mine working, communicating with the DFPlayer. Maybe I can give u a hand on that if its an "entry-Level"-Problem