PowerBroker2/DFPlayerMini_Fast #23: Setting the volume is not working correctly

targetarchitecture, 2020-11-09:
Setting the volume to 10 and then retrieving the volume shows it has been set to 24, the sound is also not reduced in volume. I'm using an ESP32 with hardware serial port on non-default pins (16,4). I have used other libraries (DFRobotDFPlayerMini) and they work setting the volume. Any chance you could have a look. Thanks John.

`#include <Arduino.h>
#include <DFPlayerMini_Fast.h>

DFPlayerMini_Fast myMP3;

#define DFPLAYER_RX 16
#define DFPLAYER_TX 4

void setup()
{
  Serial.begin(115200); //ESP32 USB Port
  Serial.flush();

  Serial1.begin(9600, SERIAL_8N1, DFPLAYER_RX, DFPLAYER_TX);

  myMP3.begin(Serial1,750);

  if (!myMP3.begin(Serial1))
  {
    Serial.println("DFPlayer error");

    delay(1000);
    ESP.restart();
  }

  myMP3.volume(10);
  delay(30);

  int16_t v = myMP3.currentVolume();

  Serial.print("Volume:");
  Serial.println(v);

  int16_t fileCounts = myMP3.numSdTracks();

  if (fileCounts == -1)
  {
    Serial.println("DFPlayer no files found");

    delay(1000);
    ESP.restart();
  }
  else
  {
    Serial.print("Found ");
    Serial.print(fileCounts);
    Serial.println(" files");
  }

  //play the mp3
  myMP3.play(random(1, 10));
}

void loop()
{
}
`

PowerBroker2, 2020-11-10:
Try putting a 10ms delay before the `volume()` call. Also, it never ceases to amaze me how many people are interested in querying the DFPlayer for the volume level that they already are setting in the code - why not simply keep the value in a variable and use the variable instead of querying? Either way, I've used the volume feature of the current library release without problems, so I'm not sure what could be wrong. Any additional details would be very helpful.

targetarchitecture, 2020-11-10:
I like to check it has been actioned. I fried my dfplayer today putting onto my board in the wrong pins. I have ordered some new ones.

PowerBroker2, 2020-11-11:
> I like to check it has been actioned
I understand that - I'm saying that once you have it working, there's no reason to check

Either way, if you find a specific bug, I'd be happy to fix it

targetarchitecture, 2020-11-12:
Thanks PB2 - your suggestion worked. I like your code and I'm using it on my daughters project boards....

https://github.com/targetarchitecture/pxt-sn3

closed 2020-11-12 by targetarchitecture

Infi-power, 2021-01-23:
I'm also having this issue as well.

I edited the example code a bit for volume testing, but it didn't worked as expected, when I try to print current volume of the DFPlayer board from serial monitor it prints -1. This issue never happened on `DFRobotDFplayerMini` library.

And this is the current code:

```c++
#include <DFPlayerMini_Fast.h>

#if !defined(UBRR1H)
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
#endif

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);

#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial);
#else
  Serial.begin(9600);
  myMP3.begin(Serial1);
#endif

 
  delay(20);
  myMP3.volume(10);
  
  int16_t v = myMP3.currentVolume();
  Serial.println(v);
  
  myMP3.play(1);
  
}

void loop()
{
  //do nothing
}
```

I'm using Arduino Nano currently.