PowerBroker2/DFPlayerMini_Fast #24: function isPlaying not working.

timothydillan, 2020-11-11:
Hey there. Thanks a lot for the library!

I currently have a project that needs to queue a music automatically when an RFID card is tapped. When the song currently playing is finished, the next music in queue will be played.

I tried using the DFRobotDFPlayerMini library, used the readType function to check whether the DFPlayer has finished playing, but the results are not accurate, making the program fail.

When googling around, I found out that your library has a isPlaying function that should check whether a song is currently being played, and so I tried using your library, it worked fine in terms of playing music, but the isPlaying function isn't working (always returns false / 0) even though a music is currently playing.

Here's my sketch:
```
LiquidCrystal_I2C lcd(0x27, 16, 2);

bool firstPlay = false;

QueueDFPlayer QMusic;

HardwareSerial hwSerial(1);

DFPlayerMini_Fast myMP3Player;

const char* wifiName = “SSID”;
const char* wifiPass = “PASS”;

RFID rfid(SS_PIN, RST_PIN);

void setup() {

  hwSerial.begin(9600, SERIAL_8N1, DFPLAYER_TX, DFPLAYER_RX);  // speed, type, TX, RX
  Serial.begin(115200);

  Serial.println();
  Serial.println(F("DFRobot DFPlayer"));
  Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));

  SPI.begin();
  rfid.init();

  pinMode(LED_PIN, OUTPUT);

  lcd.clear();

  lcd.begin();

  lcd.backlight();

  if (!myMP3Player.begin(hwSerial)) {
    Serial.println(F("Unable to begin:"));
    Serial.println(F("1.Please recheck the connection!"));
    Serial.println(F("2.Please insert the SD card!"));
    while (true);
  }

  Serial.println(("DFPlayer Mini online."));

  myMP3Player.setTimeout(500); //Set serial communictaion time out 500ms

  myMP3Player.volume(30);

  Serial.println();

  Serial.print("Connecting to ");
  Serial.println(wifiName);

  lcd.setCursor(1, 0);
  lcd.print("Initialization");
  lcd.setCursor(2, 1);
  lcd.print("Connecting...");

  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(wifiName, wifiPass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());   //You can get IP address assigned to ESP
}

void loop() {

  if (ModeAlat != "SCAN")
    return;

  rfid.halt();

  digitalWrite(LED_PIN, HIGH);
  delay(500);
  digitalWrite(LED_PIN, LOW);
  
  int tPlayMusic = 0;

  Serial.print("Is Playing State: ");
  Serial.println(myMP3Player.isPlaying());

  if (!myMP3Player.isPlaying()) {
    tPlayMusic = QMusic.getMusic();
    if (tPlayMusic != 0)
      myMP3Player.playFromMP3Folder(tPlayMusic);
  } else {
    Serial.print("Is Playing State: ");
    Serial.println(myMP3Player.isPlaying());
  }

  if (!rfid.isCard() || !rfid.readCardSerial()) {
    lcd.setCursor(0, 0);
    lcd.print("System Available");
    lcd.setCursor(0, 1);
    lcd.print("Scan your card>>");
    return;
  }

  digitalWrite(LED_PIN, HIGH);

  Serial.println("Card found");

  String RFID = String(rfid.serNum[0], HEX) + "-" + String(rfid.serNum[1], HEX) + "-" + String(rfid.serNum[2], HEX) + "-" + String(rfid.serNum[3], HEX) + "-" + String(rfid.serNum[4], HEX);

  tone(Buzzer, NOTE_E7, 100, BUZZER_CHANNEL);

  Serial.print("RFID: ");
  Serial.println(RFID);
  Serial.println("");

  String host = hostSCAN;
  host += "&rfid=";
  host += RFID;
  HTTPClient https;
  Serial.print("Request Link:");
  Serial.println(host);
  Serial.println("");

  https.begin(host);

  int httpCode = https.GET();            //Send the GET request
  String payload = https.getString();    //Get the response payload from server

  https.end();

  Serial.print("Response Code:"); //200 is OK
  Serial.println(httpCode);       //Print HTTP return code
  Serial.println("");

  Serial.print("Returned data from Server: ");
  Serial.println(payload);    //Print request response payload
  Serial.println("");

  if (httpCode != 200) {
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("Dismissal System");
    lcd.setCursor(0, 1);
    lcd.print("HTTP: " + httpCode);
    return;
  }

  DynamicJsonDocument doc(1024);

  auto error = deserializeJson(doc, payload);

  if (error) {
    Serial.print(F("deserializeJson() failed with code "));
    Serial.println(error.c_str());
    return;
  }

  String responStatus = doc["status"].as<String>();
  String responKeterangan = doc["ket"].as<String>();
  String responNama = doc["nama"].as<String>();
  String responMusic = doc["music_file"].as<String>();

  Serial.println();
  Serial.print("Status: ");
  Serial.println(responStatus);

  Serial.print("Keterangan: ");
  Serial.println(responKeterangan);

  Serial.print("Name: ");
  Serial.println(responNama);

  Serial.print("Music File No.: ");
  Serial.println(responMusic);

  Serial.println("-------------------");
  Serial.println();

  tone(Buzzer, NOTE_E7, 100, BUZZER_CHANNEL);
  delay(80);
  noTone(Buzzer, BUZZER_CHANNEL);
  tone(Buzzer, NOTE_E7, 100, BUZZER_CHANNEL);

  if (responStatus != "success") {
    lcd.clear();
    lcd.print("Please register");
    lcd.setCursor(2, 1);
    lcd.print("an account.");
    return;
  }

  lcd.clear();
  lcd.setCursor(0, 0);
  lcd.print("Please wait.....");
  lcd.setCursor(0, 1);
  lcd.print("Calling " + responNama);

  int music_file = atoi(responMusic.c_str());
  
  Serial.print("Music retrieved from database: ");
  Serial.println(music_file);

  int playMusic = 0;
  if (!myMP3Player.isPlaying()) {
    QMusic.addMusic(music_file);
    playMusic = QMusic.getMusic();
    if (playMusic != 0)
      myMP3Player.playFromMP3Folder(playMusic);
  } else {
    Serial.print("Is Playing State: ");
    Serial.println(myMP3Player.isPlaying());
    QMusic.addMusic(music_file);
  }    
  Serial.println("Scan another card.");

  delay(1000);
  
}
```

PowerBroker2, 2020-11-12:
You could insert a 10ms delay before each `isPlaying()` call, however, it would be much more reliable to monitor the digital output pin `Busy` to see if a track is currently being played

timothydillan, 2020-11-13:
> You could insert a 10ms delay before each `isPlaying()` call, however, it would be much more reliable to monitor the digital output pin `Busy` to see if a track is currently being played

Yeah, I decided to read the busy pin instead. Thanks.

Closing this issue as no further discussion is needed.

closed 2020-11-13 by timothydillan