PowerBroker2/DFPlayerMini_Fast #28: playFolder not working

ejmorgan55, 2021-01-27:
playFolder function is not working. All other functions such as playRandom work. I have 3 folders on the SD (01, 02, 03) and 112 MP3s in each folder (0001 to 0112). Fat 32 file, all formatted correctly IAW DFplayer instructions. Am I mission something? I simplified the code for inclusion below:

#include <arduino.h>
#include <DFPlayerMini_Fast.h>
#if !defined(UBRR1H)
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX
#endif
DFPlayerMini_Fast myMP3;
const byte pinDfpBusy = 12;  

void setup()
{
  Serial.begin(115200);
  Serial.begin(9600);
 
#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial);
#else
  Serial.begin(9600);
  myMP3.begin(Serial1);
#endif
  
  myMP3.volume(20);
  delay(5); 

  myMP3.playFolder (1,2)
}

void loop()
{
  
  }

harryberlin, 2021-01-27:
you do begin Serial twice.
no Serial1

ejmorgan55, 2021-01-27:
I eliminated Serial 1; albeit, I use it in the longer program. Eliminating serial1 did not help. I still cannot get it to play any folder. It will play random; all, or individual MP3s; however, it will not play any folder request? Any troubleshooting I can do?

PowerBroker2, 2021-01-27:
The following sketch works for me on a Mega:

```C++
#include <DFPlayerMini_Fast.h>

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(9600);
  
  myMP3.begin(Serial1);
  
  myMP3.volume(20);
  delay(20); 

  myMP3.playFolder(1,2);
}

void loop()
{
  
}
```

If you need SoftwareSerial, it shouldn't be too hard to replace Serial1 manually.

ejmorgan55, 2021-01-27:
Thanks; however, this script does not compile on a Nano?

   Serial
exit status 1
'Serial1' was not declared in this scope

If I remove serial1, it will compile; however, it will not playFolder. I do appreciate the help though.

PowerBroker2, 2021-01-27:
> If you need SoftwareSerial, it shouldn't be too hard to replace Serial1 manually.

```C++
#include <DFPlayerMini_Fast.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX
DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  
  myMP3.begin(mySerial);
  
  myMP3.volume(20);
  delay(20); 

  myMP3.playFolder(1,2);
}

void loop()
{
  
}
```

PowerBroker2, 2021-01-27:
Also, make sure to follow the filename rules as specified in the readme:

> If the files are played from folders other than root or "mp3", the folder names must be zero-padded 2-digit numbers while their contents must be audio files with names prepended by a zero-padded 3-digit number.

ejmorgan55, 2021-01-28:
Thanks; I had removed the serial1 as in the program above; however, as stated, the dfplayer does not playFolder. If I replace playFolder with, for example, randomAll, it plays fine. I have checked the mp3 protocol several times. I have an folder in root named mp3; in the mp3 folder, I have 3 folders named 01,02,03. In each folder, I have mp3 files named 0001, 0002, 0003, etc; should work. I have used the dfplayer in the past with an arduino nano without problems constructing a music box. Not sure why I cannot get the playFolder to work on this one. It seems unlikely; however, I wonder if it is a problem with the dfplayer chip? Regardless, thanks for taking time to help. If you have other suggestions, please let me know.

ejmorgan55, 2021-01-28:
OK; so I pulled the folders out of the folder named mp3; and simply pasted them into the root directory as 01,02,03, and it now works? Not sure why they will not play via the mp3 folder directory, unless I have been misreading the data sheet for the dfplayer?

PowerBroker2, 2021-01-28:
> I have 3 folders named 01,02,03. In each folder, I have mp3 files named 0001, 0002, 0003, etc; should work.

No, if you look at the readme excerpt, each mp3 needs a 3 digit, zero-filled number. The function does work if the file structure and code is setup properly

closed 2021-01-29 by PowerBroker2