PowerBroker2/DFPlayerMini_Fast #29: myMp3.Play() works but myMp3.Loop() dosn't

oOJoshOo, 2021-01-28:
myMp3.Play() works but myMp3.Loop() at the same place dosn't

PowerBroker2, 2021-01-29:
It works for me (just tested on my Mega).

Can you post your code in tags along with a detailed description of the file structure and filename list on your SD card?

oOJoshOo, 2021-01-31:
#define PIXEL_PIN    6 

#define PIXEL_COUNT 14

int Opening_Time = 900; // time the saber need to start in milliseconds

int t = Opening_Time/ PIXEL_COUNT;

const int BUTTON_PIN = 2; // the number of the pushbutton pin
const int SHORT_PRESS_TIME = 800; // 1000 milliseconds
const int LONG_PRESS_TIME  = 800; // 1000 milliseconds
const int Extra_LONG_PRESS_TIME  = 3000;

int Blynk_Off;
int Button_Count;
int runs;
// Variables will change:
int lastState = HIGH;  // the previous state from the input pin
int currentState;     // the current reading from the input pin
unsigned long pressedTime  = 0;
unsigned long releasedTime = 0;

int flow = 0;
#define BLYNK_PRINT Serial

#include <DFPlayerMini_Fast.h>
DFPlayerMini_Fast myMP3;

#include <SoftwareSerial.h>
SoftwareSerial SerialBLE(10, 11); // RX, TX
SoftwareSerial SerialSound(7, 8); // RX, TX

#include <BlynkSimpleSerialBLE.h>
#include <Adafruit_NeoPixel.h>
#include <EEPROM.h>

Adafruit_NeoPixel strip(PIXEL_COUNT, PIXEL_PIN, NEO_GRB + NEO_KHZ800);

char auth[] = "6vtgUge2l7Dl0RkgiOcYTLOGciKwZGTK";

int R;
int G;
int B;
int S;

BLYNK_WRITE(V0) {
  int r = param[0].asInt();
  int g = param[1].asInt();
  int b = param[2].asInt();
  static1(r, g, b);
}
BLYNK_WRITE(V1) {
  int stable = param[0].asInt();
  static2(stable);
}

void static2(int stable) {
S = stable;
}

void static1(int r, int g, int b) {
R = r;
G = g;
B = b;
}

//////////////////////////////////SETUP////////////////////////////////////////
void setup()
{

  //Serial.begin(9600);

  SerialSound.begin(9600);
  myMP3.begin(SerialSound);
  delay(20);
  myMP3.volume(30);
  
  
   R = EEPROM.read(0);
   G = EEPROM.read(1);
   B = EEPROM.read(2);
   S = EEPROM.read(3);

  
  
  pinMode(BUTTON_PIN, INPUT_PULLUP);

  strip.begin();
  strip.show();
  

  
}

////////////////LOOP/////////////////////////////////////////////////////////////////
void loop()
{

  currentState = digitalRead(BUTTON_PIN);

  if(lastState == HIGH && currentState == LOW)        // button is pressed
    pressedTime = millis();
  else if(lastState == LOW && currentState == HIGH) { // button is released
    releasedTime = millis();
  }
    long pressDuration = releasedTime - pressedTime;
    
///////////////////////////Lichtschwert Öffnen//////////////////////////////////  

    if( pressDuration < SHORT_PRESS_TIME && pressDuration > 0){
         releasedTime = 0;
         pressedTime = 0;
         EEPROM.update(0, R);
         EEPROM.update(1, G);
         EEPROM.update(2, B);
         EEPROM.update(3, S);
         
         Button_Count = ++Button_Count;
    }

    if( Button_Count == 1){
        while(flow < PIXEL_COUNT){
        strip.setPixelColor(flow, R, G, B);
        strip.show();
        delay(t);
        flow= ++flow;
       }
     runs =1;
     Button_Count = 2;
     }
//////////////////Instabil///////////////////////////////////
    if(runs == 1){ 
      myMP3.loop(2);
      if(S == 1){
         int ra = random(0, PIXEL_COUNT);
         int rb = random(0, PIXEL_COUNT);
         int rc = random(0, PIXEL_COUNT);  
           strip.setPixelColor(ra, R/4, G/4, B/4);
           strip.setPixelColor(rb, R/4, G/4, B/4);
           strip.setPixelColor(rc, R/4, G/4, B/4);
           strip.show();
           delay(60);
           strip.setPixelColor(ra, R, G, B);
           strip.setPixelColor(rb, R, G, B);
           strip.setPixelColor(rb, R, G, B);
           strip.show();
      }
    }

     
///////////////////Blaster Schuss///////////////////////////    
    if( Button_Count == 3){
    int r = random(0, PIXEL_COUNT);
    int rn= r++;
    strip.setPixelColor(r, 255, 255, 255);
    strip.setPixelColor(rn, 255, 255, 255);
    strip.show();
    delay(100);
    strip.setPixelColor(r, R, G, B);
    strip.setPixelColor(rn, R, G, B);
    strip.show();
    Button_Count = 2;
    }

//////////////////Lichtschwert Scließen//////////////////////////////////////////////
    
    if( pressDuration > LONG_PRESS_TIME && pressDuration < Extra_LONG_PRESS_TIME){
       while(flow >= 0){
        strip.setPixelColor(flow, 0, 0, 0);
        strip.show();
        delay(t);
        flow = --flow;      
       }
      Button_Count = 0;
      runs = 0; 
    }
  

  // save the the last state
  lastState = currentState;

  
 ////////////////Blynk Anschalten//////////////////////
 
 if( pressDuration > Extra_LONG_PRESS_TIME ){
  Blynk_Off= ++Blynk_Off;
  releasedTime = 0;
  pressedTime = 0;
 }

  if(Blynk_Off == 1 ){
  SerialBLE.begin(9600);
  Blynk.begin(SerialBLE, auth);
  Serial.println("Waiting for connections...");
  Blynk_Off= 2;
 }
  if(Blynk_Off > 1 ){
  Blynk.run();

  }
  
  // You can inject your own code or combine it with other sketches.
  // Check other examples on how to communicate with Blynk. Remember
  // to avoid delay() function!
  // read the state of the switch/button:
}

oOJoshOo, 2021-01-31:
This is the full code to this day. But it's not finished yet

the file structure is: SD/mp3/0001Opening.mp3
                                              0002Humming.mp3
                                              0003Closing.mp3

oOJoshOo, 2021-01-31:
Its running on an arduino nano

PowerBroker2, 2021-01-31:
Do you have a simpler example that I can run to test?

oOJoshOo, 2021-01-31:
Not yet, tomorrow I'll be working on a simpler sketch.

oOJoshOo, 2021-02-07:
Sry for the late answer, i have the same problem with your example scetch. The myMp3.play works but the loop dosent

`#include <DFPlayerMini_Fast.h>

#if !defined(UBRR1H)
#include <SoftwareSerial.h>
SoftwareSerial mySerial(7, 8); // RX, TX
#endif

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);

#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial);
#else
  Serial1.begin(9600);
  myMP3.begin(Serial1);
#endif
  
  Serial.println("Setting volume to max");
  myMP3.volume(30);
  delay(20);
  
  Serial.println("Playing track 1 for 5 sec");
  myMP3.play(1);
  delay(5000);

  Serial.println("Sleeping for 5 sec");
  myMP3.sleep();
  delay(5000);

  Serial.println("Waking up");
  myMP3.wakeUp();

  Serial.println("Looping track 1");
  myMP3.loop(2);
}

void loop()
{
  //do nothing
}`

oOJoshOo, 2021-02-07:
I found out, thet if I play track 2 and loop track 1 everything works.
could it may be that there is a length limit for loop?

oOJoshOo, 2021-02-07:
ok track 2 doesn't want to loop even if it is exactly as long as track 1

oOJoshOo, 2021-02-07:
here are the sounds i'm using
[Sounds.zip](https://github.com/PowerBroker2/DFPlayerMini_Fast/files/5940106/Sounds.zip)

PowerBroker2, 2021-02-08:
delete these lines and see if it helps:

```C++
  Serial.println("Sleeping for 5 sec");
  myMP3.sleep();
  delay(5000);

  Serial.println("Waking up");
  myMP3.wakeUp();
```

oOJoshOo, 2021-02-08:
Unfortunately, that didn't fix the problem either.
In addition, these lines are not present in my actual code.

PowerBroker2, 2021-02-16:
This works perfectly for me, as long as I'm not using your mp3 files:

```C++
#include <DFPlayerMini_Fast.h>

#if !defined(UBRR1H)
#include <SoftwareSerial.h>
SoftwareSerial mySerial(7, 8); // RX, TX
#endif

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);

#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial);
#else
  Serial1.begin(9600);
  myMP3.begin(Serial1);
#endif
  
  Serial.println("Setting volume to max");
  myMP3.volume(30);
  delay(20);
  
  Serial.println("Playing track 1 for 5 sec");
  myMP3.play(1);
  delay(5000);

  Serial.println("Looping track 1");
  myMP3.loop(2);
}

void loop()
{
  //do nothing
}
```

I think the humming track is too small for the DFPlayer to recognize or isn't in the right format for some reason. I'd suggest using clips that are at least a second long AND use wav files. Using wav files will decrease the amount of time in between calling `play()` or `loop()` and the sound actually playing (mp3 files need to be decompressed before playing unlike wavs).

I'd love to see the lightsaber once you're done with it - cool project idea.

PowerBroker2, 2021-02-18:
OP unresponsive, assuming issue is resolved - please reopen if not

closed 2021-02-18 by PowerBroker2