PowerBroker2/DFPlayerMini_Fast #31: numTracksInFolder return value if folder empty

raffaeletani, 2021-02-09:
what is the intended return value if the folder does not contain any mp3?

for testing purposes, i left the folder 05 empty, but numTracksInFolder(5) return 6 somehow.

i'm using the following code to keep track of how many songs are in each playlist:

    const int playlistLimit = 10; 
    int16_t  folderCount[playlistLimit];

    for (byte i = 0; i < playlistLimit; i += 1) {
        folderCount[i] = DFP.numTracksInFolder(i+1); 
        Serial.print(F("Playlist "));
        Serial.print(i+1);
        Serial.print(F(" contains"));
        Serial.print(folderCount[i]);
        Serial.println(F(" songs"));
    }

What am i missing?

PowerBroker2, 2021-02-16:
Looks like I didn't realize this query returns an error if the following response query is returned: `7E FF 06 40 00 00 06 FE B5 EF`. The 6 in the 4th from last byte of the error is being interpreted as "6 tracks". I'll see about fixing it asap

PowerBroker2, 2021-02-16:
Once I tag it out, version 1.2.1 will have your fix - thank you!

closed 2021-02-16 by PowerBroker2