PowerBroker2/DFPlayerMini_Fast #35: Ardiuno Code with such library 'not compiling'.

Gemma281, 2021-03-17:
This is my code that I got off of instructables;

#include <DFPlayerMini_Fast.h>
#include <Wire.h>
#include <SoftwareSerial.h>

//-----Declare MP3
SoftwareSerial mp3Serial(2,3); // RX, TX
DFPlayerMini_Fast mp3;
byte mp3vol;

//-----Constant bytes
const byte potiIN = A7;
const byte pirIN = A6;

//###################################
//###################################

void setup() {
  pinMode(potiIN, INPUT);
  pinMode(pirIN, INPUT);

  Serial.begin(9600);
  Wire.begin();

  //Start mp3 and set volume from Poti
  mp3Serial.begin(9600);
  mp3.begin(mp3Serial);
  mp3vol = map(analogRead(potiIN), 0, 1023, 0, 30);
  mp3.volume(mp3vol);
}

//================================

void loop() {
  //Set volume from poti
  mp3vol = map(analogRead(potiIN), 0, 1023, 0, 30);
  mp3.volume(mp3vol);

  //     -->                                                          Adjust the number of tracks on your SD card here: random(1, YOURVALUE);
  if(analogRead(pirIN)>5 && !mp3.isPlaying()) mp3.playFromMP3Folder(random(1, 10));  //Alternative:   random(1, mp3.numSdTracks()+1));
  
  delay(2000);
}

Can you possibly help me realize why or suggest reasons why the code wouldn't be compiling properly, I have tried reinstalling Ardunio, this library.

Incase you want more context to what it is for project wise I am basing it off of this:
https://www.instructables.com/Relaxing-Sounds-BOX-motion-Sensitive-3D-Printed/

Thanks.

PowerBroker2, 2021-03-17:
Please post the compiler errors in tags and I'll see what's going wrong

Gemma281, 2021-03-17:
Actually I figured it out a few minutes after I posted the issue and I feel stupid now, I didn't realise that it relies on fire timer.h so I downloaded that and added it to my code and all good and compiling properly, just now not downloading onto board! grrr.

closed 2021-03-17 by PowerBroker2