PowerBroker2/DFPlayerMini_Fast #38: Question about performance

fabiuz7, 2021-03-22:
Hi @PowerBroker2 , 
thank you for your work. I would ask which optimization you had applied to improve execution speed, as claimed in the description. Just curiosity :) About simplicity I have no doubt, it is clearer than others.

PowerBroker2, 2021-03-22:
This library is faster than the original since the original has a 8ms or so hard-coded delay for every command (and possibly a longer one for queries). This delay is not hard-coded in this lib. Are you having issues with processing speed?

fabiuz7, 2021-03-22:
Actually I have a project that uses the original library, but I never did test on performance neither looked into the library.
I have just discovered your library, I was wondering a migration to your solution.

Are those hardcoded delays useful to something? I had already meet other libraries making use of "casual" delays...

PowerBroker2, 2021-03-22:
> Are those hardcoded delays useful to something?

It really depends. The original lib uses the delay to keep the user from spamming commands/queries (otherwise the DFPlayer will become unresponsive), but this is only an edge case and such delays can be manually handled in the sketch if really needed.

Personally, I think my lib performs much better than the original as long as you know its limitations and how to use it.

fabiuz7, 2021-03-23:
Perfect, thanks for the clear answer. Maybe you could add a similar explanation in the readme, so new comers will immediately understand the major improvements

closed 2021-03-23 by fabiuz7