PowerBroker2/DFPlayerMini_Fast #40: Volume command not working

tomforwood, 2021-04-06:
I can't get the volume command on my dfplayer to work.
My current main loop is
`void loop() {
  static unsigned long timer = millis();
  if (millis() - timer > 3000) {
    timer = millis();
     delay(30);
     myDFPlayer.volume(5);
     mySerial.flush();
     delay(500);
     Serial.println(myDFPlayer.currentVolume());
     Serial.println(F("playing next"));
    myDFPlayer.randomAll();  //Play next mp3 every 3 second.
  }`

It does play a random different mp3 every few seconds but at full volume. Also the call to currentVolume (inserted as diagnostic) returns -1.

Also if I replace the call to randomAll() with playNext() nothing plays.

My files are /mp3/001.mp3, /mp3/002.mp3, /mp3/003.mp3

PowerBroker2, 2021-04-06:
The following works for my DFPlayer:

```C++
#include <DFPlayerMini_Fast.h>

#if !defined(UBRR1H)
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
#endif

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);

#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial, true);
#else
  Serial1.begin(9600);
  myMP3.begin(Serial1, true);
#endif
}

void loop()
{
  static unsigned long timer = millis();
  if (millis() - timer > 3000) {
    timer = millis();
    delay(30);
    myMP3.volume(5);
    delay(500);
    Serial.println(myMP3.currentVolume());
    Serial.println(F("playing next"));
    myMP3.randomAll();  //Play next mp3 every 3 second.
  }
}
```

2 things:
1) What chipset does your DFPlayer use? Some of the chips used in DFPlayer clones do not work correctly.
2) Try the sketch I posted AND post a printout of all the debug statements that result from the test

tomforwood, 2021-04-07:
Thank you for your help.

The chipset is the MH2024K which now I google I can see people online having problems with.

I am using a Mega board which I think does have UBRR!H defined but I am using software serial so I commented out the hardware serial bits of the code above

The debug output is as follows
```
09:44:14.507 -> Sent Stack:
09:44:14.507 -> 7E FF 6 6 0 0 5 FE F0 EF
09:44:14.507 -> 
09:44:15.021 -> Sent Stack:
09:44:15.021 -> 7E FF 6 43 0 0 0 FE B8 EF
09:44:15.021 -> 
09:44:15.209 -> timeout error
09:44:15.209 -> 
09:44:15.209 -> -1
09:44:15.209 -> playing next
09:44:15.256 -> Sent Stack:
09:44:15.256 -> 7E FF 6 18 0 0 0 FE E3 EF
09:44:15.256 -> 
09:44:17.495 -> Sent Stack:
09:44:17.495 -> 7E FF 6 6 0 0 5 FE F0 EF
09:44:17.495 -> 
09:44:18.005 -> Sent Stack:
09:44:18.005 -> 7E FF 6 43 0 0 0 FE B8 EF
09:44:18.005 -> 
09:44:18.236 -> timeout error
09:44:18.236 -> 
09:44:18.236 -> -1
09:44:18.236 -> playing next
09:44:18.236 -> Sent Stack:
09:44:18.236 -> 7E FF 6 18 0 0 0 FE E3 EF
09:44:18.236 -> 
09:44:20.509 -> Sent Stack:
09:44:20.509 -> 7E FF 6 6 0 0 5 FE F0 EF
09:44:20.509 -> 
```

tomforwood, 2021-04-07:
I just tried it on the hardware serial and got no sound at all and got the output
`09:56:31.946 -> Sent Stack:
09:56:31.946 -> 7E FF 6 6 0 0 5 FE F0 EF
09:56:31.946 -> 
09:56:32.389 -> Sent Stack:
09:56:32.389 -> 7E FF 6 43 0 0 0 FE B8 EF
09:56:32.389 -> 
09:56:32.481 -> Rec: 0
09:56:32.481 -> State: find_start_byte
09:56:32.527 -> Rec: 0
09:56:32.527 -> State: find_start_byte
09:56:32.527 -> Rec: 0
09:56:32.527 -> State: find_start_byte
09:56:32.527 -> Rec: 0
09:56:32.527 -> State: find_start_byte
09:56:32.573 -> Rec: 0
09:56:32.573 -> State: find_start_byte
09:56:32.573 -> Rec: 0
09:56:32.573 -> State: find_start_byte
09:56:32.573 -> timeout error
09:56:32.573 -> 
09:56:32.573 -> -1
09:56:32.573 -> playing next
09:56:32.620 -> Sent Stack:
09:56:32.620 -> 7E FF 6 18 0 0 0 FE E3 EF
09:56:32.620 -> 
09:56:34.971 -> Sent Stack:
09:56:34.971 -> 7E FF 6 6 0 0 5 FE F0 EF
09:56:34.971 -> 
09:56:35.408 -> Sent Stack:
09:56:35.408 -> 7E FF 6 43 0 0 0 FE B8 EF
09:56:35.408 -> 
09:56:35.502 -> Rec: 0
09:56:35.502 -> State: find_start_byte
09:56:35.502 -> Rec: 0
09:56:35.502 -> State: find_start_byte
09:56:35.548 -> Rec: 0
09:56:35.548 -> State: find_start_byte
09:56:35.548 -> Rec: 0
09:56:35.548 -> State: find_start_byte
09:56:35.548 -> Rec: 0
09:56:35.548 -> State: find_start_byte
09:56:35.595 -> timeout error
09:56:35.595 -> 
09:56:35.595 -> -1
09:56:35.595 -> playing next
09:56:35.595 -> Sent Stack:
09:56:35.595 -> 7E FF 6 18 0 0 0 FE E3 EF
09:56:35.595 -> 
09:56:37.875 -> Sent Stack:
09:56:37.875 -> 7E FF 6 6 0 0 5 FE F0 EF
09:56:37.875 -> 
09:56:38.385 -> Sent Stack:
09:56:38.385 -> 7E FF 6 43 0 0 0 FE B8 EF
09:56:38.385 -> 
09:56:38.477 -> Rec: 0
09:56:38.477 -> State: find_start_byte
09:56:38.477 -> Rec: 0
09:56:38.477 -> State: find_start_byte
09:56:38.522 -> Rec: 0
09:56:38.522 -> State: find_start_byte
09:56:38.522 -> Rec: 0
09:56:38.522 -> State: find_start_byte
09:56:38.567 -> Rec: 0
09:56:38.567 -> State: find_start_byte
09:56:38.567 -> Rec: 0
09:56:38.567 -> State: find_start_byte
09:56:38.567 -> timeout error
09:56:38.567 -> 
09:56:38.567 -> -1
09:56:38.567 -> playing next
09:56:38.567 -> Sent Stack:
09:56:38.567 -> 7E FF 6 18 0 0 0 FE E3 EF
09:56:38.567 -> 
09:56:40.887 -> Sent Stack:
09:56:40.887 -> 7E FF 6 6 0 0 5 FE F0 EF
09:56:40.887 -> 
09:56:41.396 -> Sent Stack:
09:56:41.396 -> 7E FF 6 43 0 0 0 FE B8 EF
09:56:41.396 -> 
09:56:41.489 -> Rec: 0
09:56:41.489 -> State: find_start_byte
09:56:41.489 -> Rec: 0
09:56:41.489 -> State: find_start_byte
09:56:41.489 -> Rec: 0
09:56:41.489 -> State: find_start_byte
09:56:41.534 -> Rec: 0
09:56:41.534 -> State: find_start_byte
09:56:41.534 -> Rec: 0
09:56:41.534 -> State: find_start_byte
09:56:41.582 -> Rec: 0
09:56:41.582 -> State: find_start_byte
09:56:41.582 -> timeout error
09:56:41.582 -> 
`

PowerBroker2, 2021-04-07:
What type of Arduino are you using? I think your DFPlayer is the problem here tbh. [These work much better since they aren't clones](https://www.amazon.com/gp/product/B089D5NLW1/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)

closed 2021-04-08 by PowerBroker2

tomforwood, 2021-04-09:
I have it attached to a Elegoo mega2560 at the moment.
I guess this is what you get if you buy chead knockoffs and don't read the reviews.
Unfortunately I'm in the UK and UK amazon doesn't seem to have the original versions