PowerBroker2/DFPlayerMini_Fast #44: Checksum calculatation

enjoyneering, 2021-05-27:
I've tried to use you library and failed. For example playFromMP3Folder() doesn't work. The problem is in findChecksum(). To fix the problem:

1. Replace all sendStack.paramLSB = 1; with sendStack.paramLSB = 0; in DFPlayerMini_Fast.cpp
2. Change findChecksum() code to this

`void DFPlayer::findChecksum(stack& _stack)
{
  int16_t checksum = 0 - (_stack.version + _stack.length + _stack.commandValue + _stack.feedbackValue + _stack.paramMSB + _stack.paramLSB);

  _stack.checksumMSB = checksum >> 8;
  _stack.checksumLSB = checksum & 0x00FF;
}`

PowerBroker2, 2021-05-27:
Interesting, thank you for bringing this to my attention. What chip is on your DFPlayer? Are you using a true DFRobot module or a clone?

enjoyneering, 2021-05-27:
I have this module
[image]

It would also be nice to have track #0 protection.. For example:
``void DFPlayer::playFromMP3Folder(uint16_t trackNum)
{
  if (trackNum == 0) trackNum = 1;

  sendStack.commandValue  = dfplayer::USE_MP3_FOLDER;
  sendStack.feedbackValue = dfplayer::NO_FEEDBACK;
  sendStack.paramMSB      = trackNum >> 8;
  sendStack.paramLSB      = trackNum & 0x00FF;

  findChecksum(sendStack);
  sendData();
}

PowerBroker2, 2021-05-27:
> It would also be nice to have track #0 protection. For example:

I know this is a terrible thing to say, but I'm going to trust the user to know not to do this.

Also, I added the other suggestions you had to the master branch - can you confirm it's working for you now? Once validated, I'll tag out a new version.

enjoyneering, 2021-05-27:
All good.

Also your library depends on FireTimer.h, which is too heavy for me. I've replaced it with two private functions & one private variable:

.h
``
private:
	uint32_t _thresholdStart;
        uint16_t _threshold;

        void _timerStart();
        bool _timerFire();

cpp
``
void DFPlayer::_timerStart()
{
  _thresholdStart = millis();
}

bool DFPlayer::_timerFire()
{
  return _threshold >= (millis() - _thresholdStart);
}

closed 2021-05-28 by enjoyneering