PowerBroker2/DFPlayerMini_Fast #46: DFPlayer mini with different chip

tsctrl, 2021-06-12:
Hi,

i have multiple version dfplayer mini with different decoder chip, when using with this library the other than the original DFplayer can receive tx command (plau/pause/stop) but did not transmit on rx such as volume and track number. is there any way i could debug the code and insert new type of byte code to use with different DF player version? how can i start?

thanks @PowerBroker2

agfaps, 2021-11-01:
Hi @tsctrl ,
Some chip can respond nicely to command without checksum.
Try comment LSB and MSB checksum in senddata() function.
If not, like MH2024K-16SS (16 pin) chip, then you need to find datasheet that explain the serial command to this chip.
Hope this helps.

derniwi, 2023-06-21:
Hi,

sometimes a "delay" command between to commands is helpfull to give the player time to process your request.

ForrestErickson, 2024-09-30:
I made this code to work on a DFPlayer with TD5580A chip

```
/*
   File: DFPlayerTD5580ATest
   Author: Forrest Lee Erickson
   Date: 20240929
*/

#define COMPANY_NAME "pubinv.org "
#define PROG_NAME "DFPlayerTD5580ATest "
#define VERSION "V0.1 "
#define DEVICE_UNDER_TEST "PMD: LCD and DFPlayer"  //A PMD model number
#define LICENSE "GNU Affero General Public License, version 3 "
#define ORIGIN "USA"

#define BAUDRATE 115200
#define BAUD_DFPLAYER 9600

#include <DFRobotDFPlayerMini.h>

// Create a SoftwareSerial object with RX and TX pins
//SoftwareSerial mySerial(16, 17); // RX, TX

HardwareSerial mySerial1(2);  // Use UART2
DFRobotDFPlayerMini dfPlayer;

const int LED_PIN = 2; // ESP32 LED
const int LED_D6 = 23; // ESP32 GPIO23 pin 16 on kit.

const int trac1 = 1;
const int trac2 = 2;

//Functions
void setupDFPlayer() {
  //  Setup UART for DFPlayer
  Serial.println("UART2 Begin");
  mySerial1.begin(BAUD_DFPLAYER, SERIAL_8N1, 16, 17);
  while (!mySerial1) {
    ; // wait for DFPlayer serial port to connect.
  }
  // Essential Initialize of DFPlayer Mini
  Serial.println("DFPlayer begin with ACK and Reset");
  dfPlayer.begin(mySerial1, true, true); // (Stream &stream, bool isACK, bool doReset)

//  delay(30);
//  Serial.println("Reset DFPlayer.");
  dfPlayer.reset();
  delay(3000);
  Serial.println("Begin DFPlayer again.");
  if (!dfPlayer.begin(mySerial1, true, true)) {
    Serial.println("DFPlayer Mini not detected or not working.");
    while (true); // Stop execution
  }
  Serial.println("DFPlayer Mini detected!");

  delay(3000);  //Required for volum to set
  //  // Set volume (0 to 30)
//  dfPlayer.volume(30);       // Set initial volume max
  dfPlayer.volume(1);       // Set initial volume low
  delay(3000);
  Serial.print("Volume is set to: ");
  digitalWrite(LED_D6, HIGH); //Start of volume read.
  Serial.println(dfPlayer.readVolume());           //Causes program lock up  
  digitalWrite(LED_D6, LOW); //End of volume read.
  
  dfPlayer.setTimeOut(500);  // Set serial communictaion time out 500ms
  delay(100);

 
  //  dfPlayer.EQ(0);          // Normal equalization //Causes program lock up  
  
}//setupDFPLayer

void setup() {
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  pinMode(LED_D6, OUTPUT);
  digitalWrite(LED_D6, LOW); //Start out low

  // Start serial communication
  Serial.begin(BAUDRATE);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }
  delay(500);
  Serial.println("===================================");
  Serial.println(DEVICE_UNDER_TEST);
  Serial.print(PROG_NAME);
  Serial.println(VERSION);
  Serial.print("Compiled at: ");
  Serial.println(F(__DATE__ " " __TIME__) ); //compile date that is used for a unique identifier
  Serial.println("===================================");
  Serial.println();

  setupDFPlayer();

  Serial.println("End of setup");
  digitalWrite(LED_PIN, LOW);
}// end of setup()

void loop() {
  Serial.println("Play a track");
  dfPlayer.play(1);
  delay(2000);
  Serial.println("Play MP3 folder");
  dfPlayer.playMp3Folder(1);
  delay(2000);

}// end of loop()
```

MatheusCaltran, 2025-03-17:
@ForrestErickson I tested it here using DFPlayer with TD5580A but it doesn't recognize the chip

15:31:25.799 -> UART2 Begin
15:31:25.799 -> DFPlayer begin with ACK and Reset
15:31:29.482 -> Begin DFPlayer again.

What version of the DFRobotDFPlayerMini library are you using?