PowerBroker2/DFPlayerMini_Fast #48: .volume()  function not working at all, always set to the highest level

supermandt, 2021-06-28:
I have tried 5 different brands of mp3 dfmini players and now just in the past few weeks, I have noticed that after uploading to the arduino and playing sounds, the sounds is always at the highest volume. The volume() function is almost ignored now.  I even set delays before and after setting the volume and nothing, always plays any mp3 at the highest level distorting sound. Just a few weeks ago, been using .volume(24) and working fine.  Even tried setting the volume to 10, still no change in volume.  Any ideas what is going on with the DFPLAYER MINI FAST library?  Does not seem to matter which mp3 player, the volume is always maxed out, so it looks to be a software bug.  Thank you!

PowerBroker2, 2021-06-28:
Can you post the sketch you're using and the accompanying debug printout? Make sure to run the sketch with the library in debug mode.

supermandt, 2021-06-28:
void setup() {
  Serial.begin(9600);
  mySerial.begin(9600);
  myMP3.begin(mySerial);
  myMP3.volume(20);
  delay(20);

Also,    myMP3.currentVolume() reports back as 29 if I do not set the volume.

And later in code, all I am doing is this:
        myMP3.playFromMP3Folder(nTrack);  

Hope this helps?
thanks.!

PowerBroker2, 2021-06-28:
Please post the whole sketch and the whole debug output. I can't tell much without those

To set debug mode, do

```
myMP3.begin(mySerial, true);
```

supermandt, 2021-06-28:
All I have is a button that when pressed sets off the sounds.  Its been working over a year the same code, but up until the last update or two, no volume is working.   Here is my sketch:

#include "Arduino.h"
#include "SoftwareSerial.h"
#include <DFPlayerMini_Fast.h>

#define SPEAKER_VOLUME 22

#define RX 10
#define TX 11
SoftwareSerial mySerial(RX, TX); // RX, TX
DFPlayerMini_Fast myMP3;

uint8_t touchState = 0;
#define TOUCH_BUTTON1 8

void setup() {
  Serial.begin(9600);

  mySerial.begin(9600);
  myMP3.begin(mySerial);
  myMP3.volume(SPEAKER_VOLUME);
  delay(20);
 
  pinMode(RX, INPUT);
  pinMode(TX, OUTPUT);
  pinMode(TOUCH_BUTTON1,INPUT);

    
  
}

void loop()
{

   touchState=digitalRead(TOUCH_BUTTON1);

   if(touchState == HIGH)
   {
        if(myMP3.isPlaying())
        {
          Serial.println("Sorry, music is already playing");
          myMP3.stop();
          delay(750);
        }
        else
        {

          myMP3.playFromMP3Folder(1);  

        }
   }
}

PowerBroker2, 2021-06-28:
I'm glad to help, but I can't if I don't have all the right info. Thank you for the sketch, though.

Again, please change the library's class constructor arguments to the following:

```
myMP3.begin(mySerial, true);
```

and post the debug printouts from the serial monitor

supermandt, 2021-06-28:
thank you for your help... here is debug
Sent Stack:
7E FF 6 6 0 0 16 FE DF EF

Sent Stack:
7E FF 6 42 0 0 0 FE B9 EF

Rec: 7E
State: find_start_byte
Rec: FF
State: find_ver_byte
Rec: 6
State: find_len_byte
Rec: 42
State: find_command_byte
timeout error
PLAYING TRACK # 105
Sent Stack:
7E FF 6 12 0 0 69 FE 80 EF

supermandt, 2021-06-28:
Also the weird thing is that when I upload to arduino and test, the sounds is at the correct level.  Then I unplug and plug in again to usb port on my computer and the sound is as loud as ever again.  The only way around this now is to add a volume control module to adjust the volume.

supermandt, 2021-06-28:
and here is another debug output:

Sent Stack:
7E FF 6 42 0 0 0 FE B9 EF

Rec: 7E
State: find_start_byte
Rec: FF
State: find_ver_byte
Rec: 6
State: find_len_byte
Rec: 42
State: find_command_byte
timeout error
Sent Stack:
7E FF 6 43 0 0 0 FE B8 EF

Rec: 7E
State: find_start_byte
Rec: FF
State: find_ver_byte
Rec: 6
State: find_len_byte
Rec: 43
State: find_command_byte
timeout error
PLAYING TRACK # 79
Sent Stack:
7E FF 6 12 0 0 4F FE 9A EF

PowerBroker2, 2021-06-29:
The volume command packet bytes look good - should work. DFPlayers are insanely finikey and poorly designed. Cheap Chinese clones don't help.

What is the latest version of the lib that works for you? What were you running before the update that broke everything?

PowerBroker2, 2021-06-29:
Later I'll get around to double checking the checksums

supermandt, 2021-06-29:
I believe the last good version was two versions ago from the current version.
The weird thing is that all the DFMINI PLAYERS I purchased from Amazon, 99% of them work perfect with the volume it’s only happened in the last week or two that the volume is not working.  Thanks again for all your help.

supermandt, 2021-06-29:
Actually my code is been the same for the last year and I just noticed this like I said about two weeks ago that the volume was not working as when I was playing an MP3 it was always playing at the loudest.

supermandt, 2021-06-29:
I received a message from the mp3 manufacturer...check this out, Can we send a command in your mp3 library?

We are sorry to cause inconvenience to you.
Due to the shortage of chip materials, we have replaced the chip. so the program will change.

Please try the following two methods to adjust the volume

1. Refer to picture 1, after wiring, long press S1 or S2 to try
 
2. Refer to picture 2, connect the USB to ttl module and send the command to adjust the volume to try.

7E FF 06 04 00 00 00 FF DD EF

We will update the complete information soon and send it to you for your reference, please be patient.
[image]
[image]

supermandt, 2021-07-06:
Any luck on this?

PowerBroker2, 2021-07-06:
Sorry, been busy with personal things and whatnot. I'm looking into it rn - hopefully I can find something.

PowerBroker2, 2021-07-06:
I can't for the life of me figure out why it the library doesn't work for you other than the chips used for DFPlayers are PoS. The serial output for each command fits the datasheet and wiki specs, so idk.

As a shot in the dark, I compared the current version with the version 2 iterations ago ([found here](https://github.com/PowerBroker2/DFPlayerMini_Fast/compare/1.2.2...1.2.4)), reverted the substantive differences, and reduced the dynamic memory usage. Hopefully this works.

Try the master branch and let me know if it improves things.

supermandt, 2021-07-06:
Thank you for helping, i really appreciate it.  I tried the master and recompiled and still no good.  When i upload the code to arduino nano, it actually works. But as soon as I unplug it and plug it in again to give the circuit power, it totally ignores the VOLUME setting.  I think I will just use an addtional volume control for my projects as your library is perfect the way it is and it is obviously not your fault, it is the chip manufacters!   Again, thank you so much for your help.  If I hear any word from the engineer from the chip company, I will post here too.

harryberlin, 2021-07-06:
the table in sheet says 16bit params, that means 2 byte. may this is the point.

PowerBroker2, 2021-07-06:
> the table in sheet says 16bit params, that means 2 byte. may this is the point.

The lib already uses 2 byte parameters, but thanks!

closed 2021-07-06 by PowerBroker2