PowerBroker2/DFPlayerMini_Fast #49: DFPlayer mini volume level in all new chipsets

supermandt, 2021-08-20:
Hi PowerBroker, I just wanted to share what I figured out with the volume setting.  With all these new chipsets coming out, I was unable to set the volume level in software and which you helped me work on a few months ago.  I solved the issue, just increased the delay after the mp3.begin() in the setup  from my 20ms to 1000 and now the volume is working again!    Just wanted to share and say thank you for all your help too!

void setup() {
  Serial.begin(9600);
  mySerial.begin(9600);
  myMP3.begin(mySerial);
  delay(1000);
  myMP3.volume(SPEAKER_VOLUME);
}

PowerBroker2, 2021-08-20:
Thanks for sharing! What chipsets have you tested this with?

supermandt, 2021-08-20:
My pleasure!
All MP3 DFplayers seem to work except the MP3-TF10P(something like that)  labeled ones.

closed 2021-08-21 by PowerBroker2