PowerBroker2/DFPlayerMini_Fast #51: Sound isn't looping seamlessly

neryash, 2021-09-06:
I have this very simple code:
```
#include <SoftwareSerial.h>
#include <DFPlayerMini_Fast.h>

SoftwareSerial mySerial(10, 11); // RX, TX
DFPlayerMini_Fast myDFPlayer;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(9600);
  
  myDFPlayer.begin(mySerial);

  myDFPlayer.volume(30); //set volume low
  myDFPlayer.setTimeout(2000);
  myDFPlayer.loop(3); //loop mp3 file with leading identifier "0002"
}

void loop()
{

}
```
The sound stops after each time for 100ms or so, and then loops again, how can I make it loop seamlessly?

PowerBroker2, 2021-09-09:
If it successfully loops, any delay is caused by the MP3 chip and not the library. Using .wav files instead of .mp3 files might help decrease latency

neryash, 2021-09-09:
> If it successfully loops, any delay is caused by the MP3 chip and not the library. Using .wav files instead of .mp3 files might help decrease latency

It's already a .wav, does the side of the SD card matter? Or maybe decreasing the butrate should help?

PowerBroker2, 2021-09-09:
Idk if either of those will help, but you can try and see if they help

neryash, 2021-09-10:
Non of this worked, why can't it just loop perfectly?

PowerBroker2, 2021-09-10:
These MP3 chips are from China and kinda suck - sorry I can't help with this

closed 2021-09-10 by PowerBroker2