PowerBroker2/DFPlayerMini_Fast #55: Example not working for me

terrypin999, 2022-01-24:
What should I hear when I run your example please?
[https://github.com/PowerBroker2/DFPlayerMini_Fast/blob/master/examples/example/example.ino](url)

I expected a constant repetition of the first track on my SD card, 0001.mp3, but I get nothing.

Meanwhile I made the minor edit shown below, adding two Play commands to the previously empty loop, and reducing the volume to 5 instead of 30. Why does that not change the volume? And not play track 2?

Is there another example that reliably delivers audio and perhaps shows more functions please?

`#include <DFPlayerMini_Fast.h>

#if !defined(UBRR1H)
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
#endif

DFPlayerMini_Fast myMP3;

void setup()
{
  Serial.begin(115200);

#if !defined(UBRR1H)
  mySerial.begin(9600);
  myMP3.begin(mySerial, true);
#else
  Serial1.begin(9600);
  myMP3.begin(Serial1, true);
#endif

  delay(1000);

  Serial.println("Setting volume to max");
  myMP3.volume(5);

  Serial.println("Looping track 1");
  myMP3.loop(1);

}

void loop()
{
  myMP3.play(1);
  delay(100);

  while (myMP3.isPlaying())
  {
    //do nothing
  }

  myMP3.play(2);
  while (myMP3.isPlaying())
  {
    //do nothing
  }

}`

derniwi, 2023-06-21:
Hi, depending on the DFPlayer mini hardware which may handle something a little bit different, but the example will play
01\001.mp3
or
01\001 my song.mp3

Depending on the file system structure and your used operating system, you also should check your micro SD card and remove all (and also hidden) folders created by the OS which are not needed. I.E. Windows creates a folder "System Volume Information". Also a tool for sorting the files and folders directly on the SD card (the operating system will sort everything while displaying but this is not automatically done while copying files to an media) might be helpfull. I use "Fat Sorter" from "Holostek".

terrypin999, 2023-06-22:
Thanks, but early in the 18 months since my post I abandoned this library and successfully used the original DFR library.